Heerkog/MicroPythonBLEHID #2: Can't Reconnect

Fedack, 2021-09-17:
When I attempt to reconnect a keyboard that's already been paired with Windows 10, I get this:
Started advertising
Central connected:  0
Central disconnected

Connecting for the first time seems to be fine, but as soon as I do a reset, it doesn't work.

Heerkog, 2021-09-17:
This is odd. You're mentioning a reset. Is this a full power cycle on the keyboard? If so, the Bluetooth might require pairing again. 

Note that this package does not (yet?) support Bluetooth bonding.

Fedack, 2021-09-17:
What's happening is that it pairs fine the first time, then any time I do a
d.start() it requires forgetting it in Windows Bluetooth then adding it
again.

I see, well that's unfortunate. Is there any intention to support it
because aside of that, everything works superbly and was easy to implement.
That is literally the only thing that'd prevent usage. For a peripheral
being unable to pair once again is a requirement at this point.
That said, it seems to be ATTEMPTING to. Windows tries to pair again but it
seems the device rejects the pairing, then it tries again and again, it
goes in a pairing loop.

Heerkog, 2021-09-17:
I'm assuming you are using my keyboard example and that you're calling d.stop() before calling d.start() for the second time. 

The way this simple keyboard example works is that in the init, it calls self.keyboard.start(), where keyboard is the device from the package. When you call d.stop(), it calls self.keyboard.stop(), which stops all Bluetooth services. When you then call d.start(), it never calls self.keyboard.start() again. If you want this start/stop behavior, I suggest implementing a d.restart() method that first calls  calls self.keyboard.start() and then calls the d.start() loop.

I hope this solves the issue!

Fedack, 2021-09-17:
The issue I am experiencing is where I reset the device (by pressing the button or unplugging) then it tries to start again but doesn't succeed in reconnecting to Windows.

Heerkog, 2021-09-17:
Ok, thanks. I think this is an issue where (somehow) Windows doesn't recognize the device as the same one that was paired earlier. I'm unsure why this is happening. It could be the lack of bonding, but I'm unsure. It could also be another Bluetooth issue. I would need to look into this, but can't promise quick results. As a temporary solution, you could consider starting/stopping the device and putting the device in some deep sleep mode instead of fully power cycling.

Fedack, 2021-09-17:
Thank you for your answers, I'll try to fiddle with it. Hopefully you figure it out.

Heerkog, 2021-09-22:
I have been able to replicate your issue. The issue appears when the bluetooth device is restarted either through a hardware reset or by the software calling `stop()`, creating a new device, and calling `start()` for the newly created device.

The issue is indeed a pairing/bonding issue. as confirmed through [this](https://github.com/micropython/micropython/issues/5767) MicroPython issue.

I have pushed a new version of the package, for which bonding and secure pairing can be enabled, that _should_ work. I say _should_, because the ESP32 implementation of MicroPython does NOT support this behavior (as confirmed in the linked issue). I own only ESP32 devices and, therefore, can't test properly. Calling methods related to bonding, or secure pairing, on an ESP32 will result in errors. In case your hardware is not an ESP32 device, you may try the code from my latest push and call `set_bonding(True)` and `set_le_secure(True)` before calling `start()` from the package. Please do let me know about the results! In case your device is also an ESP32 device, it seems we will both have to wait until the behavior is supported in a future MicroPython build (if ever).

Fedack, 2021-09-22:
Well, that's unfortunate since my device is an esp32.
Thank you for your hard work. I'll keep it in mind for the future when I
select hardware for ble.
Do you reckon esp8266 have the same issue?

Heerkog, 2021-09-22:
The MicroPython Bluetooth implementation notes the following under the [Bonding section](https://docs.micropython.org/en/latest/library/bluetooth.html#pairing-and-bonding):

> This is currently only supported when using the NimBLE stack on STM32 and Unix (not ESP32).

This is all I know.

Fedack, 2021-09-22:
Alright, thank you. Good thing this was a personal project and nothing
more. Ultimately it's just an annoyance but can be worked around.
I'm still surprised something as bad was left in. Esp32 is super popular
and mostly because of its Bluetooth. Oh, wait esp8266 doesn't even have
Bluetooth so that answers that question.

jd3096-mpy, 2022-07-26:
Now esp32 micropython supported Pairing and bonding section,I have tested,it worked.
But esp32 is still can't auto reconnect paired device after power cycling device.
How to make it work?

Heerkog, 2022-07-26:
> Now esp32 micropython supported Pairing and bonding section,I have tested,it worked.

Do you have a source for this information? The documentation still says it's not supported for ESP32. 

> But esp32 is still can't auto reconnect paired device after power cycling device. How to make it work?

Call `mouse.set_bonding(True)` before `mouse.start()`.

jd3096-mpy, 2022-07-26:
> Do you have a source for this information? The documentation still says it's not supported for ESP32.
>
>
> Call `mouse.set_bonding(True)` before `mouse.start()`.

Thanks here is link
https://github.com/micropython/micropython/issues/7008

Heerkog, 2022-07-26:
Nice! Four days ago.

Can you try calling `_ble.gap_pair(self.conn_handle)` on [line 243](https://github.com/Heerkog/MicroPythonBLEHID/blob/9e3b273e05bebd4ee71ad241705786c2f394ddc7/hid_services.py#L243) and let us know if that works?

jd3096-mpy, 2022-07-26:
> _ble.gap_pair(self.conn_handle)

I test it with android phone,
I add code on line 243,it works
everytime esp32 restart, the phone remind me to pair
but the normal status is pair at the first time ,auto connected later
I try to Call set_bonding(True) before start() but it doesn't work
still disconnected instantly

```
Server created
get secret:  10 0 b'irk'
BLE on
Registering services
Writing service characteristics
Writing device information service characteristics
Writing battery service characteristics
Writing hid service characteristics
Advertiser created:  Mouse  with services:  [UUID(0x1812)]
Server started
Started advertising
Central connected:  1
encryption update 1 1 0 1 16
```

Heerkog, 2022-07-26:
I'm going to assume that the `_IRQ_GATTS_READ_REQUEST` returns an event stating '_GATTS_ERROR_INSUFFICIENT_ENCRYPTION', `_GATTS_ERROR_INSUFFICIENT_AUTHENTICATION` or something similar.

After calling `set_bonding(True)`, can you call `set_le_secure(True)`?

You can set i/o capabilities by calling `set_io_capability(capability)` if you want. The standard value for this is `_IO_CAPABILITY_NO_INPUT_OUTPUT` with passkey 1234.

Heerkog, 2022-07-29:
@jd3096-mpy did this work?

jd3096-mpy, 2022-08-21:
> @jd3096-mpy did this work?

passkey works 
but it still can't connected auto after reboot

Heerkog, 2024-08-06:
@Fedack @jd3096-mpy 

I have just pushed a version which allows bonding, solving this issue.
Note that it requires MicroPython 1.20+ (tested with 1.23).

closed 2024-08-06 by Heerkog

jd3096-mpy, 2024-08-06:
> @Fedack @jd3096-mpy
> 
> I have just pushed a version which allows bonding, solving this issue. Note that it requires MicroPython 1.20+ (tested with 1.23).

Thanks a lot.
I test with MicroPython 1.23, it works.

wenwen19115, 2024-09-04:
[image]

wenwen19115, 2024-09-04:
This issue occurs again on MicroPython version 1.24, and the Bluetooth that has been paired before cannot be connected after restarting. Steps: Restart ESP32 and run the keyboard_example.py to print the following data:
 Server created
get secret:  2 0 None
get secret:  10 0 b'irk'
set secret:  (10, b'irk') b'\xb0\xd4\xa5|7R\xfd\xef\xa8\x0f\xf3\x9b\xafN\xc3\x08'
BLE on
Registering services
Writing service characteristics
h_mod = 16 h_ser = 18 h_fwr = 20 h_hwr = 22 h_swr = 24 h_man = 26 h_pnp = 28
h_bat = 31
Writing device information service characteristics
Writing battery service characteristics
Writing hid service characteristics
Advertiser created:  Keyboard  with services:  [UUID(0x1812)]
Server started
Started advertising
Central connected:  0
get secret:  1 0 b'\x00\x16\xa7\xebR\xd2\xd4'
Central disconnected:  0
Started advertising
Central connected:  0
get secret:  1 0 b'\x00\x16\xa7\xebR\xd2\xd4'
Central disconnected:  0
Started advertising
Central connected:  0
get secret:  1 0 b'\x00\x16\xa7\xebR\xd2\xd4'
Central disconnected:  0
Started advertising
Central connected:  0
get secret:  1 0 b'\x00\x16\xa7\xebR\xd2\xd4'
Central disconnected:  0

Heerkog, 2024-09-04:
I've pushed a new version that hopefully fixes the issue.

wenwen19115, 2024-09-04:
[image]
Hello, there is a grammatical error here, a comma is missing. Also, after I added it, I didn't get an error, but I still can't connect to the paired device.
  Line 938  (h_info, h_hid, _, self.h_rep, _, h_d1, self.h_repout, _ h_d2, h_proto) = handles[2]

Heerkog, 2024-09-04:
Can you try with [this](https://github.com/Heerkog/MicroPythonBLEHID/blob/199e48c1080102e9db5da4c20eadb4d46afeecc4/hid_services.py) earlier version?

wenwen19115, 2024-09-04:
[image]
I tried the version you gave and still have the same problem. I wonder if this has something to do with my micropython version, which is 1.24
[image]

Heerkog, 2024-09-04:
Could be. I can confirm it works perfectly with version 1.23. Could you try that?

wenwen19115, 2024-09-04:
I'll give it a try，thank you！

ctshh, 2025-06-17:
I have tried this under Micropython 1.25.0 and, as per your above comments, 1.23.0 (I'm trying to build a BLE interface for my morse key, for a morse-learning app).

[The main micropython script typing the [ and ] characters on a german keyboard layout](https://pastebin.com/raw/Nz6xTK1c)
[The version of your library I use, as of today the current one](https://pastebin.com/raw/eL6kgghg)
[The keys.json that is saved on the esp after the first connect](https://pastebin.com/raw/GMnjuejL)

Connections, disconnects and re-connects from the phone to the esp32 work fine as long as the main script is not interrupted, multiple times, all good. As soon as I stop the script and re-start it, or powercycle and start the script, a connection with the same phone does not work. As for the phone, it's an iPhone 13 running iOS 18.5.
Connections to "real" keyboards, aka. typie-typie things, work fine, as do other BLE peripherals.

Is there a way to check if the keys.json is read and used after stopping and re-starting the script?

Maybe this helps in a way, and hopefully I'm not exposing any data in the keys.json I shouldn't have. Thank you for your work on this library so far, I appreciate it!
If I can help out with more debug info (I'm afraid, only from the esp side, thank you very much Apple...), please let me know.

ThomasAtBBTF, 2025-07-27:
> Connections, disconnects and re-connects from the phone to the esp32 work fine as long as the main script is not interrupted, multiple times, all good. As soon as I stop the script and re-start it, [...]

I had the same problem, and I figured out that data written to the filesystem on a ESP32S3 does not "survive" a reset or power cycle.
For fixing this I store the data in:

```
import esp32
# --- nvs paths ---
nvsname = "keys.json"
nvsdata = esp32.NVS(nvsname)
def save_data_to_nvs(key, value):
    print("-" * 20)
    value_bytes = value.encode('utf-8')
    value_size = len(value_bytes)
    print(f"save_data_to_nvs under key:{key} value_size:{value_size} bytes for {value}")
    max_size = 1024
    try:
        nvsdata.set_i32(key+"_max", max_size)
    except Exception as e:
        print(f"Error setting max_size: {e} into key:{key+"_max"} as {max_size}")
    if value_size > max_size:
        print(f"value_size > max_size {value_size} > {max_size}")
        value_size += 1024
        nvsdata.set_i32(key+"_max", value_size)
        print(f"saving {key+"_max"} as {value_size}")
    nvsdata.set_blob(key+"_data", value_bytes)
    print(f"Data saved to {key+"_data"} as {value_bytes}.")
    print("-" * 20)

def load_data_from_nvs(key) -> str:
    print("-" * 20)
    value_str = ""
    print(f"load_data_from_nvs under key:{key}")
    max_size = 1024
    try:
        max_size = nvsdata.get_i32(key + "_max") + 1024
    except Exception as e:
        print(f"Error loading max size: {e}")
    value_bytes = bytearray(max_size)
    print(f"using max_size of {max_size} bytes")
    try:
        bytes_read = nvsdata.get_blob(key+"_data", value_bytes)
        print(f"read {bytes_read} bytes")
        if bytes_read > 0:
            value_str = value_bytes[:bytes_read].decode('utf-8')
            print(f"String retrieved: '{value_str}'")
        else:
            print("Blob found, but it contained no data.")
    except Exception as e:
        print(f"Error: {e} key:{key+"_data"}")
    print("-" * 20)
    return value_str
 
```

I made it so "complicated" because I have no idea how big the secret data can become....
Using the file system does not seem to be a good choice for ESP32 boards.

Heerkog, 2025-07-28:
@ThomasAtBBTF 
Should there not be a commit?

ThomasAtBBTF, 2025-07-28:
I am not sure...
I worked a lot on your project and also changed a lot...
I am not so familiar in working in GitHub and drive a project further.
I will send you a PM so we can discuss the best way to move on.