Heerkog/MicroPythonBLEHID #4: Pimoroni tracball PIM-447 + ESP32 - how to move cursor on screen?

c0dexter, 2022-01-11:
Hi,
sorry that I bother you, my question is probably dumb, but I'm a noob in programming on micro controllers. I found your library and example of code snippets - generaly it's working, but I can see only logs when I move the trackball, but cursor on screen is not moving. I reviewed hid_services.py lib and I cannot see there any method for moving cursor. What I suppose to do, to move cursor in real?
Hi,
sorry that I bother you, my question is probably dumb, but I'm a noob in programming on micro controllers. I found your library and example of code snippets - generaly it's working, but I can see only logs when I move the trackball, but cursor on screen is not moving. I reviewed hid_services.py lib and I cannot see there any method for moving cursor. What I suppose to do, to move cursor in real?

Heerkog, 2022-01-12:
What logs do you see?

c0dexter, 2022-01-15:
Hi, have a few more problems, but at first I will show you my logs when I pair device with Android smartphone - Huwaei P20 Pro:
`Server created
no secrets available
BLE on
Registering services
Writing service characteristics
Writing device information service characteristics
Writing battery service characteristics
Writing hid service characteristics
Advertiser created:  Trackball  with services:  [UUID(0x1812)]
Server started
Started advertising
Started advertising
Started advertising
Central connected:  0
Notify with report:  (0, 4, 6, 0)
Notify with report:  (0, 4, 5, 0)
Notify with report:  (0, 4, 4, 0)
Notify with report:  (1, 4, 4, 0)
Notify with report:  (0, 4, 4, 0)
Connection update
Notify with report:  (1, 4, 4, 0)
Notify with report:  (0, 4, 4, 0)
Notify with report:  (1, 4, 4, 0)
Started advertising
Notify with report:  (0, 4, 4, 0)
Notify with report:  (1, 4, 4, 0)
Notify with report:  (0, 4, 4, 0)
Notify with report:  (1, 4, 4, 0)
Notify with report:  (0, 4, 4, 0)
Notify with report:  (0, 3, 4, 0)
Notify with report:  (0, -1, 4, 0)
Notify with report:  (0, -3, 4, 0)
Notify with report:  (0, -3, 3, 0)
Notify with report:  (0, -5, 3, 0)
Notify with report:  (0, -7, 2, 0)
Notify with report:  (0, -9, 2, 0)
Central disconnected:  0
Started advertising
Started advertising
Started advertising`

**Issues**:
1. When I try to pair Android device (above logs) - I'm able to do this only for a few seconds - I mean that I have connection only for a few seconds, then ESP32 send me "Central disconnected:  0". In mentioned a few seconds when I have connection, I'm able to use trackball and gather logs about movement info, for example: _Notify with report:  (1, 4, 4, 0)_ Smartfont indicated the ESP device as mouse, but in real I cant see a cursor.
2. When I try to find ESP Mouse device on Windows 10, I can't see it. System isn't display this device
3. I also tried to pair with laptop with Linux Mint, but I was able to do this one day only, now, I can't pair devices, maybe it's a linux system bug. When I was paired, I coudn't move a cursor on screen when I used a tracball.

Questions:
1. How can I improve paring process and establishing a stable connection as a mouse device?
2. How can I move a cursror when I have a conncection? Is there any methods like mouse.moveToLeft/Right/Up/Down(value_here)?

Thank you in advance for reply

Heerkog, 2022-01-15:
Ha! You're using my tinypico_trackball personal project! As this is a personal project, it behalves different than you'd expect. It actually uses the trackball as a joystick that doesn't self center. As a result, your cursor flies to a corner and stays there because you're not manually centering it. You really should have started from the mouse example, not a personal project.

I can explain the issues, but the simple solution is probably to change
```
            self.axes = (self.clamp(self.axes[0] + left - right), self.clamp(self.axes[1] + up - down))
```
to
```
            self.axes = (right - left, down - up)
```

If that doesn't work, please try starting from the mouse example. You can adapt that easily to use the trackball.py file.

Finally, when you power cycle your board or restart the services, always remove the paired device before connecting again.

closed 2022-01-15 by Heerkog