Heerkog/MicroPythonBLEHID #5: Testing with Mac OS

p2c2e, 2022-01-13:
Hi Heerko,

Happy New Year and Thanks for sharing your efforts on MicroPythonBLEHID with others.

I was trying to test your code with ESP-32 (clone) and MacOS (Monterey). I understand that you haven't tested this and was not your focus.

I am able to pair (connect/disconnect) etc. Though Mac seems to display a wrong Icon (though I see your code has the correct device_appearance). Did you try to try the code with MacOS at all? Would you know what could be wrong?
As a baseline, I tested your code with an Android phone and verified it works in that case.

Sudhan

Heerkog, 2022-01-13:
Hi Sudhan,

I did not test with MacOS since I do not own any Apple devices.

I found that the assignment of icons can come from different settings. MacOS may be assigning icons on the basis of the pnp information instead of the device appearance. As a result, the wrong icon might be a result from 'faking' the pnp information.

You could try to play a bit with the device appearance first and then the pnp settings. Note that you should change these settings before starting the services.

Please keep me informed of anything you discover.

Heerko

p2c2e, 2022-01-15:
Thanks for your reply. I am a novice with BLE - and upon reading some specs, it is clear the device_appearance is meant to represent the physical form factor - not the UI representation (not sure why BLE specs would care).

That said, based on your suggestion, did some debugging on the PNP - and found some issues that I have raised a pull request for.

Still NO luck with getting the stuff to work on MacOS - and to rule out Micropython implementations, I tested the code from : https://github.com/T-vK/ESP32-BLE-Mouse ; This does not work either :( 

Will continue to play around when time permits...

closed 2022-01-15 by p2c2e