Heerkog/MicroPythonBLEHID #8: Stack overflow during connecting

Runeluna, 2022-07-12:
HI! I was running the "/examples/simple/keyboard_example.py" on a ESP32-C3 devkit. It started with no problem, but had a stack overflow error when connecting to the computer.

`Server created
no secrets available
BLE on
Registering services
Writing service characteristics
Writing device information service characteristics
Writing battery service characteristics
Writing hid service characteristics
Advertiser created:  Keyboard  with services:  [UUID(0x1812)]
Server started
Started advertising
Central connected:  1
MTU exchanged:  23

***ERROR*** A stack overflow in task ble has been detected.
Core  0 register dump:
MEPC    : 0x4038073a  RA      : 0x4038c344  SP      : 0x3fc97650  GP      : 0x3fc93a00  
TP      : 0x3fca87f4  T0      : 0x4005890e  T1      : 0x74732041  T2      : 0x00000001  
S0/FP   : 0x3fc97664  S1      : 0x80000005  A0      : 0x3fc97664  A1      : 0x3c110ee8  
A2      : 0x002e6465  A3      : 0x3fc976a0  A4      : 0x00000001  A5      : 0x3fc9c000  
A6      : 0x7265766f  A7      : 0x206b6361  S2      : 0x3fc9c4e0  S3      : 0x00000001  
S4      : 0x3fcd0e48  S5      : 0x00000020  S6      : 0x00000000  S7      : 0x00000024  
S8      : 0x3fcd250c  S9      : 0x3fcd0f1c  S10     : 0x3fcd1098  S11     : 0x3fc9d000  
T3      : 0x202a2a2a  T4      : 0x524f5252  T5      : 0x452a2a2a  T6      : 0xfda9e149  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fc97650: 0x3fce0000 0x3fce0000 0x3fcc7704 0x4038e9f8 0x00000000 0x452a2a2a 0x524f5252 0x202a2a2a
3fc97670: 0x74732041 0x206b6361 0x7265766f 0x776f6c66 0x206e6920 0x6b736174 0x656c6220 0x73616820
3fc97690: 0x65656220 0x6564206e 0x74636574 0x002e6465 0x00000000 0x00000000 0x00000000 0x40000000
3fc976b0: 0x00001881 0x80000005 0x3fc9c514 0x4038dc14 0x00000000 0x00000020 0x3fcd0e48 0x00000001
3fc976d0: 0x00001881 0x80000005 0x00000000 0x4038c3c0 0x403801c2 0x80000005 0x00000000 0x403801a6
3fc976f0: 0x00000001 0x3fc976f8 0xffffffff 0x3fca074c 0x3fca074c 0x00000001 0x3fc9770c 0xffffffff
3fc97710: 0x3fcc4bf0 0x3fcc4bf0 0x00000000 0x3fc97720 0xffffffff 0x3fc97720 0x3fc97720 0x00000000
3fc97730: 0x3fc97734 0xffffffff 0x3fc97734 0x3fc97734 0x00000000 0x3fc97748 0xffffffff 0x3fc97748
3fc97750: 0x3fc97748 0x00000000 0x3fc9775c 0xffffffff 0x3fc9775c 0x3fc9775c 0x00000000 0x3fc97770
3fc97770: 0xffffffff 0x3fc97770 0x3fc97770 0x00000000 0x3fc97784 0xffffffff 0x3fc97784 0x3fc97784
3fc97790: 0x00000000 0x3fc97798 0xffffffff 0x3fc97798 0x3fc97798 0x00000000 0x3fc977ac 0xffffffff
3fc977b0: 0x3fc977ac 0x3fc977ac 0x00000000 0x3fc977c0 0xffffffff 0x3fc977c0 0x3fc977c0 0x00000000
3fc977d0: 0x3fc977d4 0xffffffff 0x3fc977d4 0x3fc977d4 0x00000000 0x3fc977e8 0xffffffff 0x3fc977e8
3fc977f0: 0x3fc977e8 0x00000000 0x3fc977fc 0xffffffff 0x3fc977fc 0x3fc977fc 0x00000000 0x3fc97810
3fc97810: 0xffffffff 0x3fc97810 0x3fc97810 0x00000000 0x3fc97824 0xffffffff 0x3fc97824 0x3fc97824
3fc97830: 0x00000000 0x3fc97838 0xffffffff 0x3fc97838 0x3fc97838 0x00000000 0x3fc9784c 0xffffffff
3fc97850: 0x3fc9784c 0x3fc9784c 0x00000000 0x3fc97860 0xffffffff 0x3fc97860 0x3fc97860 0x00000000
3fc97870: 0x3fc97874 0xffffffff 0x3fc97874 0x3fc97874 0x00000000 0x3fc97888 0xffffffff 0x3fc97888
3fc97890: 0x3fc97888 0x00000001 0x3fc9789c 0xffffffff 0x3fcd1b20 0x3fcd1b20 0x00000000 0x3fc978b0
3fc978b0: 0xffffffff 0x3fc978b0 0x3fc978b0 0x00000001 0x3fc978c4 0xffffffff 0x3fccb8e4 0x3fccb8e4
3fc978d0: 0x00000000 0x3fc978d8 0xffffffff 0x3fc978d8 0x3fc978d8 0x00000000 0x3fc978ec 0xffffffff
3fc978f0: 0x3fc978ec 0x3fc978ec 0x00000000 0x3fc97900 0xffffffff 0x3fc97900 0x3fc97900 0x00000000
3fc97910: 0x3fc97914 0xffffffff 0x3fc97914 0x3fc97914 0x00000001 0x3fc97928 0xffffffff 0x3fc9f340
3fc97930: 0x3fc9f340 0x00000000 0x3fc9793c 0xffffffff 0x3fc9793c 0x3fc9793c 0x00000000 0x3fc97948
3fc97950: 0x00000000 0x00000000 0x00000000 0x3fc97960 0xffffffff 0x3fc97960 0x3fc97960 0x00000000
3fc97970: 0x3fc97974 0xffffffff 0x3fc97974 0x3fc97974 0x00000001 0x00000001 0x00000000 0x0001ffff
3fc97990: 0x00000000 0xb33fffff 0x00000000 0x00000000 0x3fc9799c 0x00000000 0x00000000 0x00000000
3fc979b0: 0x3fc979b4 0xffffffff 0x3fc979b4 0x3fc979b4 0x00000000 0x3fc979c8 0xffffffff 0x3fc979c8
3fc979d0: 0x3fc979c8 0x00000001 0x00000001 0x00000000 0x0001ffff 0x00000000 0xb33fffff 0x00000000
3fc979f0: 0x00000000 0x3fc9de3c 0x3fc9dea4 0x3fc9df0c 0x00000000 0x00000000 0x00000001 0x00000000
3fc97a10: 0x00000000 0x00000000 0x42100554 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc97a30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: f4eb85c502489242

Rebooting...`

Heerkog, 2022-07-12:
I'm going to assume the code is registering constant key presses. 

This might be because the software is assuming pins having pull down resistors, while the hardware is using pull up resistors, or vice versa.

Heerkog, 2022-07-12:
The code will only start advertising the device (for a short while) after a pin is triggered. If this pin will remain triggered constantly, it might explain the stackoverflow. In the other cases, it sounds like those pins are never triggered and the device therefore never starts advertising.

Please disable the reading of the pins, call advertise immediately, and after it connects (and a delay of a few seconds) send some keys (using the call back function).

coinight, 2022-07-15:
> The code will only start advertising the device (for a short while) after a pin is triggered. If this pin will remain triggered constantly, it might explain the stackoverflow. In the other cases, it [...]

I have this problem too but even if I connect nothing on the pins.The core panic also happened.
self.pin_w = Pin(0, Pin.IN,pull = Pin.PULL_DOWN)
self.pin_s = Pin(1, Pin.IN,pull = Pin.PULL_DOWN)
self.pin_d = Pin(2, Pin.IN,pull = Pin.PULL_DOWN)
self.pin_a = Pin(3, Pin.IN,pull = Pin.PULL_DOWN)
By testing.this will happen when devices connect esp32c3.And noob as i am don't know how could i do to solve the problem

Heerkog, 2022-07-15:
Can you try to call co_start_test() instead of the normal start?

```
if __name__ == "__main__":
    d = Device()
    d.co_start_test()
```

Note that once this test starts, you have 30 seconds to connect to the device from your client and then have 5 seconds to select something where the device can type output (for example, notepad).

Please let me know if this also causes a stack overflow.

coinight, 2022-07-16:
> Can you try to call co_start_test() instead of the normal start?
>
> ```
> if __name__ == "__main__":
>     d = Device()
>     d.co_start_test()
> ```
>
> Note that once this test starts, you have 30 seconds to [...]

I tried but system crashed again the infomation is below

coinight, 2022-07-16:
Server created
no secrets available
BLE on
Registering services
Writing service characteristics
Writing device information service characteristics
Writing battery service characteristics
Writing hid service characteristics
Advertiser created:  Keyboard  with services:  [UUID(0x1812)]
Server started
Started advertising
Central connected:  1

***ERROR*** A stack overflow in task ble has been detected.
Core  0 register dump:
MEPC    : 0x4038071a  RA      : 0x4038c254  SP      : 0x3fc97600  GP      : 0x3fc93a00  
TP      : 0x3fca839c  T0      : 0x4005890e  T1      : 0x74732041  T2      : 0x00000001  
S0/FP   : 0x3fc97614  S1      : 0x80000005  A0      : 0x3fc97614  A1      : 0x3c110e34  
A2      : 0x002e6465  A3      : 0x3fc97650  A4      : 0x00000001  A5      : 0x3fc9c000  
A6      : 0x7265766f  A7      : 0x206b6361  S2      : 0x3fc9c4c8  S3      : 0x00000001  
S4      : 0x3fcd0b1c  S5      : 0x3fcd1ee0  S6      : 0x3fcd1ee0  S7      : 0x00000008  
S8      : 0x00000010  S9      : 0x00000000  S10     : 0x0000009b  S11     : 0x3fc9d000  
T3      : 0x202a2a2a  T4      : 0x524f5252  T5      : 0x452a2a2a  T6      : 0x00000000  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fc97600: 0x3fce0000 0x3fce0000 0x3fcc72e8 0x4038e908 0x00000000 0x452a2a2a 0x524f5252 0x202a2a2a
3fc97620: 0x74732041 0x206b6361 0x7265766f 0x776f6c66 0x206e6920 0x6b736174 0x656c6220 0x73616820
3fc97640: 0x65656220 0x6564206e 0x74636574 0x002e6465 0x00000000 0x00000000 0x00000000 0x40000000
3fc97660: 0x00001881 0x80000005 0x3fc9c4fc 0x4038db24 0x3fcd1ee0 0x3fcd1ee0 0x3fcd0b1c 0x00000001
3fc97680: 0x00001881 0x80000005 0x0000021d 0x4038c2d0 0x403801c2 0x80000005 0x0000021d 0x403801a6
3fc976a0: 0x00000001 0x3fc976a8 0xffffffff 0x3fca067c 0x3fca067c 0x00000001 0x3fc976bc 0xffffffff
3fc976c0: 0x3fcc4bf0 0x3fcc4bf0 0x00000000 0x3fc976d0 0xffffffff 0x3fc976d0 0x3fc976d0 0x00000000
3fc976e0: 0x3fc976e4 0xffffffff 0x3fc976e4 0x3fc976e4 0x00000000 0x3fc976f8 0xffffffff 0x3fc976f8
3fc97700: 0x3fc976f8 0x00000000 0x3fc9770c 0xffffffff 0x3fc9770c 0x3fc9770c 0x00000000 0x3fc97720
3fc97720: 0xffffffff 0x3fc97720 0x3fc97720 0x00000000 0x3fc97734 0xffffffff 0x3fc97734 0x3fc97734
3fc97740: 0x00000000 0x3fc97748 0xffffffff 0x3fc97748 0x3fc97748 0x00000000 0x3fc9775c 0xffffffff
3fc97760: 0x3fc9775c 0x3fc9775c 0x00000000 0x3fc97770 0xffffffff 0x3fc97770 0x3fc97770 0x00000000
3fc97780: 0x3fc97784 0xffffffff 0x3fc97784 0x3fc97784 0x00000000 0x3fc97798 0xffffffff 0x3fc97798
3fc977a0: 0x3fc97798 0x00000000 0x3fc977ac 0xffffffff 0x3fc977ac 0x3fc977ac 0x00000000 0x3fc977c0
3fc977c0: 0xffffffff 0x3fc977c0 0x3fc977c0 0x00000000 0x3fc977d4 0xffffffff 0x3fc977d4 0x3fc977d4
3fc977e0: 0x00000000 0x3fc977e8 0xffffffff 0x3fc977e8 0x3fc977e8 0x00000000 0x3fc977fc 0xffffffff
3fc97800: 0x3fc977fc 0x3fc977fc 0x00000000 0x3fc97810 0xffffffff 0x3fc97810 0x3fc97810 0x00000000
3fc97820: 0x3fc97824 0xffffffff 0x3fc97824 0x3fc97824 0x00000000 0x3fc97838 0xffffffff 0x3fc97838
3fc97840: 0x3fc97838 0x00000001 0x3fc9784c 0xffffffff 0x3fcd1704 0x3fcd1704 0x00000000 0x3fc97860
3fc97860: 0xffffffff 0x3fc97860 0x3fc97860 0x00000001 0x3fc97874 0xffffffff 0x3fccb4c8 0x3fccb4c8
3fc97880: 0x00000000 0x3fc97888 0xffffffff 0x3fc97888 0x3fc97888 0x00000000 0x3fc9789c 0xffffffff
3fc978a0: 0x3fc9789c 0x3fc9789c 0x00000000 0x3fc978b0 0xffffffff 0x3fc978b0 0x3fc978b0 0x00000000
3fc978c0: 0x3fc978c4 0xffffffff 0x3fc978c4 0x3fc978c4 0x00000001 0x3fc978d8 0xffffffff 0x3fc9f270
3fc978e0: 0x3fc9f270 0x00000000 0x3fc978ec 0xffffffff 0x3fc978ec 0x3fc978ec 0x00000000 0x3fc978f8
3fc97900: 0x00000000 0x00000000 0x00000000 0x3fc97910 0xffffffff 0x3fc97910 0x3fc97910 0x00000000
3fc97920: 0x3fc97924 0xffffffff 0x3fc97924 0x3fc97924 0x00000001 0x00000001 0x00000000 0x0001ffff
3fc97940: 0x00000000 0xb33fffff 0x00000000 0x00000000 0x3fc9794c 0x00000000 0x00000000 0x00000000
3fc97960: 0x3fc97964 0xffffffff 0x3fc97964 0x3fc97964 0x00000000 0x3fc97978 0xffffffff 0x3fc97978
3fc97980: 0x3fc97978 0x00000001 0x00000001 0x00000000 0x0001ffff 0x00000000 0xb33fffff 0x00000000
3fc979a0: 0x00000000 0x3fc9dd6c 0x3fc9ddd4 0x3fc9de3c 0x00000000 0x00000000 0x00000001 0x00000000
3fc979c0: 0x00000000 0x00000000 0x420ff43a 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc979e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: e75378a2b5917647

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x403806e6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xe3c
load:0x403ce000,len:0x6dc
load:0x403d0000,len:0x28ac
entry 0x403ce000
MicroPython v1.19.1 on 2022-06-27; ESP32C3 module with ESP32C3
Type "help()" for more information.

coinight, 2022-07-16:
if __name__ == "__main__":
    d = Device()
    d.co_start_test()
code has been changed like this.

Heerkog, 2022-07-16:
Thank you so much for testing this.

Since I do not own an ESP32C3 board and currently don't have access to my ESP32 board, I am going to have to ask you to perform a series of checks so we can get a clue as to what the problem is.

Can you check the following:
1. Does /examples/simple/keyboard_example.py produce the same issue if we run our test using REPL/rshell:
```
from keyboard_example import Device
d = Device()
d.advertise()
# connect to the device here
d.test()
```
If it does, we can exclude asyncio from causing the problem.

2. Does /examples/simple/mouse_example.py produce the same issue if we run our test as above. If it does we can exclude the keyboard specific code.
3. Does [this old version](https://github.com/Heerkog/MicroPythonBLEHID/blob/47ec9967618005e342e0d3a0a4eea2f0b261a190/hid_services.py) also produce the  same issue if we run our test as above. If it doesn't, there is an issue in the bonding and secure bonding code that I added later

coinight, 2022-07-17:
Thanks for answering our questions.
for question 1. the problem also happened 
I guess the problem is based on the risc-v struct between mpy and esp-idf.
I tested lots of packs from c-make to pure .py which were bulit for esp32 not c3.
Some of them have problems and it does't make sense (becasue it seem that .py is Compatible between esp32 series. But not)

Heerkog, 2022-07-17:
Could be indeed. I would really like to see the outcome of the 3rd test though, because that could mean I messed something up!

ChihoSin, 2022-08-04:
[image]

It's working in peripheral mode when disable some BLE feature for esp32c3.

[image]

musingsole, 2023-01-16:
I have an [ESP32-C3 board](https://learn.adafruit.com/adafruit-qt-py-esp32-c3-wifi-dev-board) and am running into stack overflow as well. I tried using the [old hid_services.py module you linked](https://github.com/Heerkog/MicroPythonBLEHID/blob/47ec9967618005e342e0d3a0a4eea2f0b261a190/hid_services.py) but still ran into a stackoverflow:

```
>>> reset()
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x403806e6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xe3c
load:0x403ce000,len:0x6dc
load:0x403d0000,len:0x28ac
entry 0x403ce000
Starting soon. In 3 seconds
Server created
Starting
BLE on
Registering services
Writing service characteristics
Writing device information service characteristics
Writing battery service characteristics
Writing hid service characteristics
Advertiser created
Server started
Started advertising
Central connected:  1
MTU exchanged

***ERROR*** A stack overflow in task ble has been detected.
Core  0 register dump:
MEPC    : 0x4038071a  RA      : 0x4038c276  SP      : 0x3fc97600  GP      : 0x3fc93a00  
TP      : 0x3fca84f4  T0      : 0x4005890e  T1      : 0x74732041  T2      : 0x00000001  
S0/FP   : 0x3fc97614  S1      : 0x80000005  A0      : 0x3fc97614  A1      : 0x3c110e34  
A2      : 0x002e6465  A3      : 0x3fc97650  A4      : 0x00000001  A5      : 0x3fc9c000  
A6      : 0x7265766f  A7      : 0x206b6361  S2      : 0x3fc9c4c8  S3      : 0x00000001  
S4      : 0x3fcd0b18  S5      : 0x3fcd0b00  S6      : 0x3fcd0b00  S7      : 0x00000008  
S8      : 0x00000010  S9      : 0x00000000  S10     : 0x000000bc  S11     : 0x3fc9d000  
T3      : 0x202a2a2a  T4      : 0x524f5252  T5      : 0x452a2a2a  T6      : 0x00000000  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fc97600: 0x3fce0000 0x3fce0000 0x3fcc72d0 0x4038e92a 0x00000000 0x452a2a2a 0x524f5252 0x202a2a2a
3fc97620: 0x74732041 0x206b6361 0x7265766f 0x776f6c66 0x206e6920 0x6b736174 0x656c6220 0x73616820
3fc97640: 0x65656220 0x6564206e 0x74636574 0x002e6465 0x00000000 0x00000000 0x00000000 0x40000000
3fc97660: 0x00001881 0x80000005 0x3fc9c4fc 0x4038db46 0x3fcd0b00 0x3fcd0b00 0x3fcd0b18 0x00000001
3fc97680: 0x00001881 0x80000005 0x0000021f 0x4038c2f2 0x403801c2 0x80000005 0x0000021f 0x403801a6
3fc976a0: 0x00000001 0x3fc976a8 0xffffffff 0x3fca067c 0x3fca067c 0x00000001 0x3fc976bc 0xffffffff
3fc976c0: 0x3fcc4c30 0x3fcc4c30 0x00000000 0x3fc976d0 0xffffffff 0x3fc976d0 0x3fc976d0 0x00000000
3fc976e0: 0x3fc976e4 0xffffffff 0x3fc976e4 0x3fc976e4 0x00000000 0x3fc976f8 0xffffffff 0x3fc976f8
3fc97700: 0x3fc976f8 0x00000000 0x3fc9770c 0xffffffff 0x3fc9770c 0x3fc9770c 0x00000000 0x3fc97720
3fc97720: 0xffffffff 0x3fc97720 0x3fc97720 0x00000000 0x3fc97734 0xffffffff 0x3fc97734 0x3fc97734
3fc97740: 0x00000000 0x3fc97748 0xffffffff 0x3fc97748 0x3fc97748 0x00000000 0x3fc9775c 0xffffffff
3fc97760: 0x3fc9775c 0x3fc9775c 0x00000000 0x3fc97770 0xffffffff 0x3fc97770 0x3fc97770 0x00000000
3fc97780: 0x3fc97784 0xffffffff 0x3fc97784 0x3fc97784 0x00000000 0x3fc97798 0xffffffff 0x3fc97798
3fc977a0: 0x3fc97798 0x00000000 0x3fc977ac 0xffffffff 0x3fc977ac 0x3fc977ac 0x00000000 0x3fc977c0
3fc977c0: 0xffffffff 0x3fc977c0 0x3fc977c0 0x00000000 0x3fc977d4 0xffffffff 0x3fc977d4 0x3fc977d4
3fc977e0: 0x00000000 0x3fc977e8 0xffffffff 0x3fc977e8 0x3fc977e8 0x00000000 0x3fc977fc 0xffffffff
3fc97800: 0x3fc977fc 0x3fc977fc 0x00000000 0x3fc97810 0xffffffff 0x3fc97810 0x3fc97810 0x00000000
3fc97820: 0x3fc97824 0xffffffff 0x3fc97824 0x3fc97824 0x00000000 0x3fc97838 0xffffffff 0x3fc97838
3fc97840: 0x3fc97838 0x00000001 0x3fc9784c 0xffffffff 0x3fcd16ec 0x3fcd16ec 0x00000000 0x3fc97860
3fc97860: 0xffffffff 0x3fc97860 0x3fc97860 0x00000001 0x3fc97874 0xffffffff 0x3fccb4b0 0x3fccb4b0
3fc97880: 0x00000000 0x3fc97888 0xffffffff 0x3fc97888 0x3fc97888 0x00000000 0x3fc9789c 0xffffffff
3fc978a0: 0x3fc9789c 0x3fc9789c 0x00000000 0x3fc978b0 0xffffffff 0x3fc978b0 0x3fc978b0 0x00000000
3fc978c0: 0x3fc978c4 0xffffffff 0x3fc978c4 0x3fc978c4 0x00000001 0x3fc978d8 0xffffffff 0x3fc9f270
3fc978e0: 0x3fc9f270 0x00000000 0x3fc978ec 0xffffffff 0x3fc978ec 0x3fc978ec 0x00000000 0x3fc978f8
3fc97900: 0x00000000 0x00000000 0x00000000 0x3fc97910 0xffffffff 0x3fc97910 0x3fc97910 0x00000000
3fc97920: 0x3fc97924 0xffffffff 0x3fc97924 0x3fc97924 0x00000001 0x00000001 0x00000000 0x0001ffff
3fc97940: 0x00000000 0xb33fffff 0x00000000 0x00000000 0x3fc9794c 0x00000000 0x00000000 0x00000000
3fc97960: 0x3fc97964 0xffffffff 0x3fc97964 0x3fc97964 0x00000000 0x3fc97978 0xffffffff 0x3fc97978
3fc97980: 0x3fc97978 0x00000001 0x00000001 0x00000000 0x0001ffff 0x00000000 0xb33fffff 0x00000000
3fc979a0: 0x00000000 0x3fc9dd6c 0x3fc9ddd4 0x3fc9de3c 0x00000000 0x00000000 0x00000001 0x00000000
3fc979c0: 0x00000000 0x00000000 0x420ff18a 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc979e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: a88fac183168950c

Rebooting...
```

musingsole, 2023-01-21:
The same code that triggered a stack overflow on my ESP32-C3 boards is running fine on my ESP32-**S3**. The S3 has 512 kB SRAM versus the C3's 400 kB, and that seems to make the difference.

Heerkog, 2023-01-21:
Thank you for trying this. This strongly indicates that the issue is a memory related! 

I will add the memory requirements to the readme.

closed 2023-01-21 by Heerkog