Heerkog/MicroPythonBLEHID #9: OSX nothing happens

ilyamordasov, 2022-07-16:
I attached the buttons, executed async mouse example, connected to Mouse device (it is exactly detected as mouse device), so I can see 
```
Notify with report:  (0, 0, 127, 0)
Notify with report:  (0, 0, 0, 0)
Notify with report:  (0, 0, 127, 0)
Notify with report:  (0, 0, 0, 0)
Notify with report:  (0, 0, 127, 0)
```
when buttons are not pressed and some opposite values when I press any button. But mouse doesn't move, doesn't call menu/submenu, nothing. Could you explain what will happen, and about what this example?

Heerkog, 2022-07-17:
Are you able to test with something other than OSX?

I'm asking because there seem to be [issues](https://github.com/Heerkog/MicroPythonBLEHID/issues/5) with Apple which are probably related to 'faking' the device information (manufacturer, version, etc).

ilyamordasov, 2022-07-17:
Yes, with android works like a charm

Heerkog, 2022-07-17:
You could try finding the (i) device information, (ii) revision information, and (iii) php information used by an Apple approved device and using the provided functions to set these values after line 37 of the async example and see if it then works.

If that doesn't work, you might need to also find (iv) the HID input report of that same device and use that. Note that changing a HID input report requires changing the code in several places, some of which is not easy to understand.

ilyamordasov, 2022-07-17:
A few years ago I was implementing usb hid device with stm32, will look at its descriptor, but do not remember did it work with OSX or not, but exactly worked with one of them ubuntu or osx.

Heerkog, 2022-07-17:
Please check [this](https://askubuntu.com/questions/766258/using-hid-compliant-mouse-like-device-in-ubuntu) thread for Ubuntu. I feel the OSX and Ubuntu issues are very related and probably lead back to issues with the pnp and device information not leading the OS to the correct/any drivers.

In case you find working device and pnp information, please do let me know so I can update the examples!

ilyamordasov, 2022-07-17:
Ok

Heerkog, 2024-08-06:
I have just pushed a major new version of this library. If anyone could test whether this issue has been resolved, that would be great.

In case it is not resolved, a wireshark pcap trace of the faulty behavior would go a long way to me being able to fix it.

randomtestfive, 2024-10-11:
I'm running Linux Mint 22, with a TinyS3 on Micropython 1.23. Running the simple keyboard example, there's no events in Wireshark after the connection. I'm also running into #2 at the same time.

Heerkog, 2024-10-12:
That is a shame. I was hoping the issue would've been solved.

Heerkog, 2025-01-27:
This has been fixed.