Heerkog/MicroPythonBLEHID #12: Fail to boot the mouse_example without cable

GKarbon, 2022-11-20:
Hi,

I run the code on ESP32 with a gyro meter, everything is fine when I just run it in IDE.

But when I write it in main.py, try to run it at boot and without cable, it won't work.

It is able to connect to computer, but there is no notification.

Heerkog, 2022-11-20:
What do you mean without cable? You mean battery powered using the battery pins? Is the device in power saving mode? Are you using the gyroscope to control the mouse movement? Is the gyroscope powered? Is the gyroscope registering movement? 

Can you try using a usb power bank or charger (anything but the pc) to power the device? This I can confirm to work.

GKarbon, 2022-11-21:
Thanks for your reply, actually I used a 18650 battery so I think power supply not a problem.

The chip is connected to computer using a USB cable, and I edit the code with Thonny. 

The program will only start successfully when it is **connected to PC  with Thonny opened**, and it is fine to remove the cable and close Thonny after the bluetooth is connected.

If it is not connected to PC, I can find and connect with bluetooth but the mouse won't move.
If it is connected to PC but without Thonny, I can find the bluetooth device but not able to connect to it.

GKarbon, 2022-11-21:
Sorry I accidentally closed the issue. 🙏

Heerkog, 2022-11-21:
Sorry I do not own an ESP-Wroon-32, and for me the device works well on Windows 10. I'm unsure how to replicate your issue.

GKarbon, 2022-11-28:
OK I raised this issue just because I have never met a bug wired like this 😂.

Thanks for your time again!

closed 2022-11-28 by GKarbon