Heerkog/MicroPythonBLEHID #13: BLE keyboard (ESP32) sends repeating of keys after a single key due to gatts_notify( )

vsupacha, 2023-02-25:
**Error scenario:**
MicroPython v1.19.1 for ESP32. 
async keyboard example.
A single key causes non-stop repeating of keys.
Defect is found to be the calling of `self._ble.gatts_notify(self.conn_handle, self.h_rep, state)` in `notify_hid_report( )`.

**Preliminary fix**
Call `Keyboard.set_keys( )` without arguments, followed by `Keyboard.notify_hid_report( )` after the original code.

Heerkog, 2023-02-25:
> A single key causes non-stop repeating of keys.

Is the key pressed (i.e., pushed and released) or held (i.e., continuously pushed down)? 
- In case of the first, the release of the key should call `Keyboard.set_keys()` with empty bits, followed by `Keyboard.notify_hid_report()` as suggested.
- - In case of the second, this is intended behavior.

vsupacha, 2023-02-26:
The key was pressed and released. So I expect that the update in the `bluetooth` module may break the previous behavior.

Heerkog, 2023-02-26:
> The key was pressed and released. So I expect that the update in the `bluetooth` module may break the previous behavior.

Do you have a source for this update? I don't see changes in the documentation.

> BTW, thanks for your code. The module helps me a lot in order to make a game controller for a PC web game.

I'm glad my code could help you! Appreciate the feedback!

closed 2024-08-06 by Heerkog