Heerkog/MicroPythonBLEHID #15: Connects and immediatelly disconnects

aguaviva, 2023-05-07:
I see the keyboard being advertised, I connect to it but immediately disconnects. I am using linux.

See the pic below for more details:

[image]

Thanks for sharing this lib, looks very well written.

Heerkog, 2023-05-07:
Could you try without secrets?

Immediate disconnection often happens when the central (i.e., Linux) is paired with the device and the device is power cycled or the HID services are restarted. Make sure to remove the device from the central's Bluetooth devices before attempting a connection.

Note that there also seem to be several issues with Linux not recognizing notifications because, I suspect, the device information characteristics do not match with actual known devices. I'm unaware of a fix for this.

Please keep me updated.

aguaviva, 2023-05-07:
Removing the paired device and adding it again fixed it. Linux doesn't depend on the information characteristics matching any known devices, it is agnostic to that and it only expects devices to follow the protocol.

The problem now is that I get out of mem errors after a while

[image]

this is how the main loop of keyboard_example.py looks like, the rest is the same.
```python
if __name__ == "__main__":
    d = Device()
    d.advertise()
    while True:
        d.test()
        print("done")
        time.sleep_ms(2000)
```

Heerkog, 2023-05-08:
> Removing the paired device and adding it again fixed it. Linux doesn't depend on the information characteristics matching any known devices, it is agnostic to that and it only expects devices to follow the protocol.

That is indeed how it should work! Yet, I am getting reports of certain distributions not accepting notifications. I'm glad it works for you.

> The problem now is that I get out of mem errors after a while

In general this happens with boards with lower amounts of memory, or when notifications follow eachother too quickly. I'd try adding a bit more sleep time/await time in the loops.

closed 2023-05-22 by Heerkog

Heerkog, 2024-08-07:
@aguaviva Bonding is now supported with the latest versions of this library and Micropython!