Heerkog/MicroPythonBLEHID #19: Incorrect descriptor flags?

QuirkyCort, 2024-08-09:
In lines 836 to 837...

```
                (UUID(0x2A4D), F_READ_NOTIFY, ((UUID(0x2908), F_READ),)),
                (UUID(0x2A4D), F_READ_WRITE, ((UUID(0x2908), F_READ),)),     
```

...the descriptor flags are set to `F_READ`, which is equivalent to `bluetooth.FLAG_READ`, which is 0x02. But that's a descriptor, and value for a descriptor read flag should be 0x01 (see https://docs.silabs.com/bluetooth/6.0.0/bluetooth-stack-api/sl-bt-gattdb-characteristic-properties and https://docs.silabs.com/bluetooth/6.2.0/bluetooth-stack-api/sl-bt-gattdb-descriptor-properties for a list of the characteristic and descriptor flags).

I've tested on 1.19.1 and 1.23.0 (both on ESP32), and found that oddly enough, a descriptor flag of 0x01 (Read) works on 1.19.1 but not on 1.23.0, while a descriptor flag of 0x02 (Write) works on 1.23.0 but not on 1.19.1. Using 0x03 (read and write) works on both.

I don't know much about Bluetooth, so I can't even begin to guess why this is so.

Heerkog, 2024-08-09:
You're correct. I found the same behavior on Micropython v1.23 and hence changed it to a 'faulty' but working value.

I will open an issue in the Micropython repository. Thanks!

Heerkog, 2024-08-09:
https://github.com/micropython/micropython/issues/15627

Heerkog, 2024-09-03:
I have updated the descriptor to `ATT_F_READ_WRITE` to be more correct.

Heerkog, 2025-01-27:
This has been fixed.

closed 2025-01-27 by Heerkog