Heerkog/MicroPythonBLEHID #20: Mouse and keyboard?

aloneguid, 2024-09-13:
Is it possible to act as a mouse and keyboard at the same time? I have managed to make it work exclusively as keyboard or exclusively as mouse, but not simultaneously. HID spec is hard to understand,  wondering if you can help please 🙏

Heerkog, 2024-09-13:
Yes, but this functionality is not offered 'out of the box'. In order to have a mouse + keyboard, you will need to subclass HumanInterfaceDevice to create a device with service and report descriptors that support mouse and keyboard functionalities.

You can follow the same structure as the Keyboard class. The resources section of the readme has a good [tutorial](https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/) on how to create a descriptor for such a multi-device.

Once you have a multi-device descriptor, your input report should match the bytes for that descriptor.

closed 2024-09-13 by Heerkog

aloneguid, 2024-09-17:
Thank you, great tutorial by the way. I have managed to create keyboard and mouse running on pico now.

Heerkog, 2024-09-17:
Excellent! A pull request with this functionality would probably be appreciated by many others!

dahanzimin, 2024-10-16:
@aloneguid Thank you for your success. Would you be willing to share how to configure the keyboard and mouse to work together? Thank you

minghao999, 2024-12-20:
I have tried many methods but cannot understand how keyboard and mouse work simultaneously. Can you share your example? Thank you.🙏