Heerkog/MicroPythonBLEHID #21: How does the ESP32 series support automatic reconnection of paired devices

dahanzimin, 2024-10-06:
Hello, may I ask if the ESP32 analog keyboard has been paired with a WIN10 computer, but after disconnecting Bluetooth, it cannot automatically reconnect
`MPY: soft reboot
no secrets available
Server created
BLE on
Registering services
Writing service characteristics
h_mod = 21 h_ser = 23 h_fwr = 25 h_hwr = 27 h_swr = 29 h_man = 31 h_pnp = 33
h_bat = 36
Writing device information service characteristics
Writing battery service characteristics
h_info = 42 h_hid = 44 h_ctrl = 46 h_rep = 48 h_d1ref = 51 h_proto = 53
Writing hid service characteristics
Advertiser created:  mouse  with services:  [UUID(0x1812)]
Server started
Started advertising
Central connected:  1
Notify with report:  (1, -100, -100, 0)
Central disconnected:  1
Started advertising
Central connected:  1
Started advertising
Central disconnected:  1
Started advertising`

Heerkog, 2024-10-06:
Which version of MicropythonBLEHID are you using?

Which version of Micropython?

dahanzimin, 2024-10-06:
MicropythonBLEHID : Master main branch
 Micropython :  v1.22.0
 Hardware : ESP32C3
The keyboard and mouse currently being tested are unable to automatically reconnect

Heerkog, 2024-10-06:
I can confirm it works perfectly with version Micropython 1.23. Please upgrade to 1.23.

closed 2024-10-06 by Heerkog