Heerkog/MicroPythonBLEHID #22: Can't Reconnect paired device

fred2nice, 2024-11-24:
Hello!

I have the same problem, the initial connection works fine, but then after a soft reboot or power off / power on under windows 11 it connects and disconnects right away ( windows 10 ), on iPhone I need to click on the keyboard inside Bluetooth settings each time.
( I need only on Windows, but I tested on iPhone for ... testing )
So I have to remove the device and add it every time.

I tried with MicroPython 1.20, 1.23 / 1.24 / 1.25 preview, same problem.

The card is a YD-ES32-23 / ESP32-S3 WROOM 1.

If you have idea ....

I tried with different options:
```
 # Create our device
 self.keyboard = Keyboard("KeyboardFred")
 # Set a callback function to catch changes of device state
 self.keyboard.set_state_change_callback(self.keyboard_state_callback)
 self.keyboard.set_bonding(True) #False
 self.keyboard.set_le_secure(True) #False
 # Start our device
 self.keyboard.start()
```

First run :
```
Server started
Device in IDLE state, launching BLE advertising
Started advertising
Central connection: 1
MTU exchanged: 23
get secret: 2 0 None
get secret: 1 0 b'\x00\x00\x00\x00\x00\x00\x00'
get secret: 2 0 b'\x00JU\x00\x1d\xa5\x10'
Connection update. Interval= 6 latency= 0 timeout= 960 status= 0
encryption update 1 1 0 1 16
secret set: (1, b'\x00JU\x00\x1d\xa5\x10') b'\x00JU\x00\x1d\xa5\x10\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00 \x00\x00\x00\x00\x00\x00D\x9c\x01A\xd24\x1a\xaf\xa0\xd4\\HE\xddF\x17 \x01\xef\x8d\xe2\x16O\xecC\r\xbf[\xdd4\xc0S\x1e\xb8\x01\x85\xbe\xa bT\xc7\x86\x9bO\xa6\xdc0\x0c\x0ev\xc1\x81\x05\x00\x00\x00\x00\x00'
get secret: 1 0 b'\x00\x00\x00\x00\x00\x00\x00'
secret set: (2, b'\x00JU\x00\x1d\xa5\x10') b'\x00JU\x00\x1d\xa5\x10\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00 \x00\x00\x00\x00\x00\x00D\x9c\x01A\xd24\x1a\xaf\xa0\xd4\\HE\xddF\x17 \x01\x92\xa2\x16\xa4e\xc3t4d\x0f\x87b\x13\x1f\x99\xca\x01\xd2\x84\ x00erT\xf5\\xe4\x8c\n\xde\x11\x8c\x83\xba\x05\x00\x00\x00\x00\x00'
get secret: 1 0 b'\x00\x00\x00\x00\x00\x00\x00'
Connection update. Interval= 6 latency= 0 timeout= 200 status= 0
```

Second run :
```
Server created
get secret: 2 0 None
get secret: 10 0 b'irk'
secret set: (10, b'irk') b"\x1b\x12\x8e(\xbe\x1a 2vk/\xd2e>'R"
BLE on
Registration services
Writing service characteristics
h_mod = 21 h_ser = 23 h_fwr = 25 h_hwr = 27 h_swr = 29 h_man = 31 h_pnp = 33
h_bat = 36
Writing device information service characteristics
Writing battery service characteristics
Writing hid service characteristics
Advertiser created: KeyboardFred with services: [UUID(0x1812)]
Server started
Device in IDLE state, BLE advertising started
Started advertising
Central connected: 1
get secret: 1 0 b'\x00JU\x00\x1d\xa5\x10'
Central disconnected: 1
Device in IDLE state, BLE advertising started
Started advertising
Central connected: 1
get secret: 1 0 b'\x00JU\x00\x1d\xa5\x10'
Central disconnected: 1
Device in IDLE state, BLE advertising started
Started advertising
Central connected: 1
get secret: 1 0 b'\x00JU\x00\x1d\xa5\x10'
Central disconnected: 1
Device in IDLE state, BLE advertising started
Started advertising
Central connected: 1
get secret: 1 0 b'\x00JU\x00\x1d\xa5\x10'
Central disconnected: 1
```

Heerkog, 2024-11-25:
Hi Fred,

Could you check whether the secret that's being returned is correct?

Also is your bluetooth device id remaining the same between boots?

fred2nice, 2024-11-29:
HI, thank you for your reply.
I tried with 3 differents windows, I think you're right there is something bad about the the secret code.
I don't know how to check the secret code. If you can give me a clue .... 
Below my code and the output.
If you think it's an hardware trouble I can take a new card ....

```
import time
from machine import Pin
from hid_services import Keyboard

class SimpleKeyboard:
    def __init__(self):
        self.device_name = "SimpleKeyboard"
        self.keyboard = Keyboard(self.device_name)
        self.keyboard.set_state_change_callback(self.keyboard_state_callback)
        
        self.advertising_started = False
        self.connected = False
        
        # Définition des boutons
        self.buttons = [
            {"name": "bt 1", "pin": Pin(2, Pin.IN, Pin.PULL_UP), "sendKey": None, "sendString": "test", "execMacro": None},
            # Ajoutez d'autres boutons ici si nécessaire
        ]
        
        self.keyboard.set_bonding(True)
        self.keyboard.set_le_secure(True)
        self.keyboard.start()
        
        # Start advertising initially
        self.start_advertising()
        
    
    def keyboard_state_callback(self):
        state = self.keyboard.get_state()
        current_time = "{:02}:{:02}:{:02}".format(*time.localtime()[3:6])
        print(f"{current_time} - Keyboard state changed: {state}")
        if state == Keyboard.DEVICE_IDLE:
            if not self.connected:
                print(f"{current_time} - Device in IDLE state, starting BLE advertising after delay")
                time.sleep(1)  # test avec un délai à 20 secondes
                self.start_advertising()
        elif state == Keyboard.DEVICE_ADVERTISING:
            print(f"{current_time} - Device is advertising")
        elif state == Keyboard.DEVICE_CONNECTED:
            print(f"{current_time} - Device is connected")
            self.connected = True
            self.advertising_started = False
        elif state == Keyboard.DEVICE_DISCONNECTED:
            print(f"{current_time} - Device disconnected, waiting before restarting advertising")
            self.connected = False
            self.advertising_started = False
            time.sleep(1)  # test avec un délai 30 secondes pour permettre à l'état de se stabiliser
            self.start_advertising()
        else:
            print(f"{current_time} - Unknown state")
    
    def start_advertising(self):
        current_time = "{:02}:{:02}:{:02}".format(*time.localtime()[3:6])
        if not self.advertising_started and not self.connected:
            try:
                print(f"{current_time} - Attempting to start advertising...")
                self.keyboard.start_advertising()
                self.advertising_started = True
                print(f"{current_time} - Started advertising")
            except AttributeError:
                print(f"{current_time} - Failed to start advertising: advertiser not initialized")
    
    def stop_advertising(self):
[... 82 lines cut ...]
Writing service characteristics
h_mod = 21 h_ser = 23 h_fwr = 25 h_hwr = 27 h_swr = 29 h_man = 31 h_pnp = 33
h_bat = 36
Writing device information service characteristics
Writing battery service characteristics
Writing hid service characteristics
Advertiser created:  SimpleKeyboard  with services:  [UUID(0x1812)]
Server started
19:17:24 - Attempting to start advertising...
Started advertising
19:17:24 - Keyboard state changed: 2
19:17:24 - Device is advertising
19:17:24 - Started advertising
19:17:24 - Current Keyboard state: 2
19:17:24 - Current Keyboard state: 2
....
19:17:24 - Keyboard state changed: 3
19:17:24 - Device is connected
Central connected:  1
get secret:  1 0 b'\x00\xc4O\x90!4h'
19:17:24 - Current Keyboard state: 3
19:17:24 - Keyboard state changed: 1
Central disconnected:  1
19:17:24 - Current Keyboard state: 1
```

Heerkog, 2024-11-30:
Hi Alex,

Can you try the version I just pushed? When you do, please do the following:
- Remove the HID device from your Windows and Iphone
- Delete the keys.json file from your board

Thank you,

fred2nice, 2024-12-08:
Hi,

Still same problem.
Could you tell me what kind of card you use, maybe the easy way should have same hardware,-)

First connection :
```
MPY: soft reboot
no secrets available
Server created
get secret:  2 0 None
get secret:  10 0 b'irk'
set secret:  (10, b'irk') b'\xb8{\t"\x08\xd7"u2\x12\xef\x93\xd70\t\xca'
BLE on
Registering services
Writing service characteristics
h_mod = 21 h_ser = 23 h_fwr = 25 h_hwr = 27 h_swr = 29 h_man = 31 h_pnp = 33
h_bat = 36
Writing device information service characteristics
Writing battery service characteristics
Writing hid service characteristics
Advertiser created:  KeyboardFred  with services:  [UUID(0x1812)]
Server started
Appareil en état IDLE, lancement de la publicité BLE
Started advertising
Central connected:  1
MTU exchanged:  23
get secret:  2 0 None
get secret:  1 0 b'\x00\x00\x00\x00\x00\x00\x00'
get secret:  2 0 b'\x00\xc4O\x90!4h'
encryption update 1 1 0 0 16
Connection update. Interval= 12  latency= 0  timeout= 960  status= 0
Read request:  57
Read request:  53
Read request:  55
Read request:  54
Read request:  48
Read request:  51
Read request:  50
Read request:  46
Read request:  44
Read request:  44
Read request:  44
Read request:  42
Read request:  36
Read request:  38
Read request:  33
Read request:  31
Read request:  31
Read request:  29
Read request:  27
Read request:  25
Read request:  23
Read request:  21
Read request:  21
Read request:  36
Connection update. Interval= 12  latency= 0  timeout= 200  status= 0
```

Second :```

MPY: soft reboot
Server created
get secret:  2 0 None
get secret:  10 0 b'irk'
set secret:  (10, b'irk') b'+j\x7f\x81\x86=\xb9a \x00\xc5\xb2\xd4a\x18\x12'
BLE on
Registering services
Writing service characteristics
h_mod = 21 h_ser = 23 h_fwr = 25 h_hwr = 27 h_swr = 29 h_man = 31 h_pnp = 33
h_bat = 36
Writing device information service characteristics
Writing battery service characteristics
Writing hid service characteristics
Advertiser created:  KeyboardFred  with services:  [UUID(0x1812)]
Server started
Appareil en état IDLE, lancement de la publicité BLE
Started advertising
Central connected:  1
get secret:  1 0 b'\x00\xc4O\x90!4h'
Central disconnected:  1
Appareil en état IDLE, lancement de la publicité BLE
Started advertising
Central connected:  1
get secret:  1 0 b'\x00\xc4O\x90!4h'
Central disconnected:  1
```

keys.json
`[[10, "aXJr\n", "K2p/gYY9uWEgAMWy1GEYEg==\n"]]`

Heerkog, 2024-12-08:
Hi Fred,

Apologies, it seems my updates weren't pushed to GitHub. Could you please take the next steps again? It should now save keys correctly and display keys consistently.

1. Remove the Bluetooth device from your pc.
2. Remove ``keys.json`` from your board.
3. Update ``hid_services.py``.

It should now save keys correctly and display keys consistently.

The board that I use is a [TinyPICO](https://www.tinypico.com/). Newer versions are available at [Unexpected Maker](https://unexpectedmaker.com/).

fred2nice, 2024-12-09:
Hi Heerkog,

I did several tries, I removed keys.json, remove the device from windows.
Same trouble, now in keys.json, I have several values

```
from hid_services import Keyboard
class Device:
    def __init__(self):
    ...............
        # Create our device
        self.keyboard = Keyboard("KeyboardFred")
        # Set a callback function to catch changes of device state
        self.keyboard.set_state_change_callback(self.keyboard_state_callback)
        # Start our device
        self.keyboard.start()

```

keys.json
```
[[1, "AMRPkCE0aA==", "AMRPkCE0aBAAAAAAAAAAAAAAAAAAAAAA3g9zIzcNBSKAgIq+K71j/QHvjeIWT+xDDb9b3TTAUx64AemhaPw91lIm6RaVJdJEdnYFAAAAAAA="], [2, "AMRPkCE0aA==", "AMRPkCE0aBAAAAAAAAAAAAAAAAAAAAAA3g9zIzcNBSKAgIq+K71j/QFfSZWG9mxL7ZSLL7SRkb/sARiKvLP5BgGwvZJW8937nGkFAAAAAAA="], [10, "aXJr", "g7U0LXKYE2XiM23RvCcYwQ=="]]
```

First Run :
    
```
MPY: soft reboot
no secrets available
Server created
get secret: 2 0 None None
get secret: 10 0 b'irk' None
set secret: (10, b'irk') b'\xbf8jY\xfd\x8dFu\xcb\x1f\xd7\xaa\x9d\xe8\x85-'
BLE on with public mac address b"<\x84'\xd8r\x1a"
Registering services
Writing service characteristics
h_mod = 21 h_ser = 23 h_fwr = 25 h_hwr = 27 h_swr = 29 h_man = 31 h_pnp = 33
h_bat = 36
Writing device information service characteristics
Writing battery service characteristics
Writing hid service characteristics
Advertiser created:  KeyboardFred  with services:  [UUID(0x1812)]
Server started
Appareil en état IDLE, lancement de la publicité BLE
Started advertising
Central connected:  1
MTU exchanged:  23
get secret: 2 0 None None
get secret: 1 0 b'\x00\x00\x00\x00\x00\x00\x00' None
get secret: 2 0 b'\x00\xc4O\x90!4h' None
encryption update 1 1 0 1 16
set secret: (1, b'\x00\xc4O\x90!4h') b'\x00\xc4O\x90!4h\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xde\x0fs#7\r\x05"\x80\x80\x8a\xbe+\xbdc\xfd\x01\xef\x8d\xe2\x16O\xecC\r\xbf[\xdd4\xc0S\x1e\xb8\x01\xe9\xa1h\xfc=\xd6R&\xe9\x16\x95%\xd2Dvv\x05\x00\x00\x00\x00\x00'
get secret: 1 0 b'\x00\x00\x00\x00\x00\x00\x00' None
set secret: (2, b'\x00\xc4O\x90!4h') b'\x00\xc4O\x90!4h\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xde\x0fs#7\r\x05"\x80\x80\x8a\xbe+\xbdc\xfd\x01_I\x95\x86\xf6lK\xed\x94\x8b/\xb4\x91\x91\xbf\xec\x01\x18\x8a\xbc\xb3\xf9\x06\x01\xb0\xbd\x92V\xf3\xdd\xfb\x9ci\x05\x00\x00\x00\x00\x00'
get secret: 1 0 b'\x00\x00\x00\x00\x00\x00\x00' None
Read request:  57
Read request:  53
```

Second run :
    
```
MPY: soft reboot
Server created
get secret: 2 0 None b'\x00\xc4O\x90!4h\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xde\x0fs#7\r\x05"\x80\x80\x8a\xbe+\xbdc\xfd\x01_I\x95\x86\xf6lK\xed\x94\x8b/\xb4\x91\x91\xbf\xec\x01\x18\x8a\xbc\xb3\xf9\x06\x01\xb0\xbd\x92V\xf3\xdd\xfb\x9ci\x05\x00\x00\x00\x00\x00'
get secret: 10 0 b'irk' b'\xab_Oe\xc3\xa1\xea\xa6{\x02\xbf!\x17"H\xe8'
set secret: (10, b'irk') b"\x83\xb54-r\x98\x13e\xe23m\xd1\xbc'\x18\xc1"
BLE on with public mac address b"<\x84'\xd8r\x1a"
Registering services
Writing service characteristics
[... 41 lines cut ...]
get secret: 1 0 b'\x00\xc4O\x90!4h' b'\x00\xc4O\x90!4h\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xde\x0fs#7\r\x05"\x80\x80\x8a\xbe+\xbdc\xfd\x01\xef\x8d\xe2\x16O\xecC\r\xbf[\xdd4\xc0S\x1e\xb8\x01\xe9\xa1h\xfc=\xd6R&\xe9\x16\x95%\xd2Dvv\x05\x00\x00\x00\x00\x00'
Central disconnected:  1
Appareil en état IDLE, lancement de la publicité BLE
Started advertising
Central connected:  1
get secret: 1 0 b'\x00\xc4O\x90!4h' b'\x00\xc4O\x90!4h\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xde\x0fs#7\r\x05"\x80\x80\x8a\xbe+\xbdc\xfd\x01\xef\x8d\xe2\x16O\xecC\r\xbf[\xdd4\xc0S\x1e\xb8\x01\xe9\xa1h\xfc=\xd6R&\xe9\x16\x95%\xd2Dvv\x05\x00\x00\x00\x00\x00'
Central disconnected:  1
Appareil en état IDLE, lancement de la publicité BLE
Started advertising
Central connected:  1
get secret: 1 0 b'\x00\xc4O\x90!4h' b'\x00\xc4O\x90!4h\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xde\x0fs#7\r\x05"\x80\x80\x8a\xbe+\xbdc\xfd\x01\xef\x8d\xe2\x16O\xecC\r\xbf[\xdd4\xc0S\x1e\xb8\x01\xe9\xa1h\xfc=\xd6R&\xe9\x16\x95%\xd2Dvv\x05\x00\x00\x00\x00\x00'
Central disconnected:  1
Appareil en état IDLE, lancement de la publicité BLE
Started advertising
Central connected:  1
get secret: 1 0 b'\x00\xc4O\x90!4h' b'\x00\xc4O\x90!4h\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xde\x0fs#7\r\x05"\x80\x80\x8a\xbe+\xbdc\xfd\x01\xef\x8d\xe2\x16O\xecC\r\xbf[\xdd4\xc0S\x1e\xb8\x01\xe9\xa1h\xfc=\xd6R&\xe9\x16\x95%\xd2Dvv\x05\x00\x00\x00\x00\x00'
Central disconnected:  1
Appareil en état IDLE, lancement de la publicité BLE
Started advertising
Central connected:  1
get secret: 1 0 b'\x00\xc4O\x90!4h' b'\x00\xc4O\x90!4h\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xde\x0fs#7\r\x05"\x80\x80\x8a\xbe+\xbdc\xfd\x01\xef\x8d\xe2\x16O\xecC\r\xbf[\xdd4\xc0S\x1e\xb8\x01\xe9\xa1h\xfc=\xd6R&\xe9\x16\x95%\xd2Dvv\x05\x00\x00\x00\x00\x00'
Central disconnected:  1
Appareil en état IDLE, lancement de la publicité BLE
Started advertising
```

Heerkog, 2024-12-09:
Hi Fred,

I'm a bit confused. Why is your connection handle 1 instead of 0? Are you trying to connect with multiple clients? HID devices may only serve a single connection.

In any case, it is clear that your client isn't accepting the secret that is returned on your second run, eventhough the secret seems to be correct. Your MAC also remains the same. The only way that I can replicate your issue, is by deleting the ``keys.json`` file between the first and second run.

Anyway, for reference, I'm putting my output below. I made some small changes for clearer output, and will push those to GitHub for you. I used a Tinypico and Windows 10. 

First run:
```
No secrets available
Server created
Returning secret: None for index 0 with type 2
Returning secret: None for key (10, b'irk') with type 10
Saving secret: (10, b'irk') b'\xd9\x15\xb0 \x96\xfb\xedPM\xb2\x89\x97\x97\xe2ZP'
Returning secret: None for index 0 with type 2
BLE on with public mac address b'\x98\xcd\xac\xd3F\xa6'
Registering services
Writing service characteristics
h_mod = 21 h_ser = 23 h_fwr = 25 h_hwr = 27 h_swr = 29 h_man = 31 h_pnp = 33
h_bat = 36
Writing device information service characteristics
Writing battery service characteristics
Writing hid service characteristics
Advertiser created:  Joystick  with services:  [UUID(0x1812)]
Server started
Started advertising
Central connected:  0
MTU exchanged:  23
Returning secret: None for index 0 with type 2
Returning secret: None for key (1, b'\x00\x00\x00\x00\x00\x00\x00') with type 1
Returning secret: None for key (2, b'\x00\x8b\x8d\xef\xcf\x08\xb8') with type 2
Encryption update 0 1 0 1 16
Saving secret: (1, b'\x00\x8b\x8d\xef\xcf\x08\xb8') b'\x00\x8b\x8d\xef\xcf\x08\xb8\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00f\xd2\xc9|O\x07\x9b\x1ad2(>\xba u\xe9\x01\xd9\x15\xb0 \x96\xfb\xedPM\xb2\x89\x97\x97\xe2ZP\x01\xc0\x84\x88}W80e\xc8:\xcc\xd0\xba\x8b\x95\x18\x05\x00\x00\x00\x00\x00'
Saving secret: (2, b'\x00\x8b\x8d\xef\xcf\x08\xb8') b"\x00\x8b\x8d\xef\xcf\x08\xb8\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00f\xd2\xc9|O\x07\x9b\x1ad2(>\xba u\xe9\x01\xb6\x1bk\xf4S\xb84E\xdb\xb2\xc5\xbc\x98\x90A.\x01u\x8c\xf6a'\x1b\x0e\xc6\x89\xa7\x92\xa9\x06\x00\xed#\x05\x00\x00\x00\x00\x00"
Connection update. Interval= 15  latency= 0  timeout= 960  status= 0
Read request:  53
Read request:  48
Read request:  51
Read request:  50
Read request:  46
Read request:  44
Read request:  44
Read request:  42
Read request:  36
Read request:  38
Read request:  33
Read request:  31
Read request:  31
Read request:  29
Read request:  27
Read request:  25
Read request:  23
Read request:  21
Read request:  21
Read request:  36
Connection update. Interval= 48  latency= 0  timeout= 960  status= 0
Connection update. Interval= 12  latency= 0  timeout= 200  status= 0
```

Second run:
```
Server created
Returning secret: b"\x00\x8b\x8d\xef\xcf\x08\xb8\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00f\xd2\xc9|O\x07\x9b\x1ad2(>\xba u\xe9\x01\xb6\x1bk\xf4S\xb84E\xdb\xb2\xc5\xbc\x98\x90A.\x01u\x8c\xf6a'\x1b\x0e\xc6\x89\xa7\x92\xa9\x06\x00\xed#\x05\x00\x00\x00\x00\x00" for index 0 with type 2
Returning secret: None for index 1 with type 2
Returning secret: b'\xd9\x15\xb0 \x96\xfb\xedPM\xb2\x89\x97\x97\xe2ZP' for key (10, b'irk') with type 10
Returning secret: b"\x00\x8b\x8d\xef\xcf\x08\xb8\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00f\xd2\xc9|O\x07\x9b\x1ad2(>\xba u\xe9\x01\xb6\x1bk\xf4S\xb84E\xdb\xb2\xc5\xbc\x98\x90A.\x01u\x8c\xf6a'\x1b\x0e\xc6\x89\xa7\x92\xa9\x06\x00\xed#\x05\x00\x00\x00\x00\x00" for index 0 with type 2
Returning secret: None for index 1 with type 2
BLE on with public mac address b'\x98\xcd\xac\xd3F\xa6'
Registering services
Writing service characteristics
h_mod = 21 h_ser = 23 h_fwr = 25 h_hwr = 27 h_swr = 29 h_man = 31 h_pnp = 33
h_bat = 36
Writing device information service characteristics
Writing battery service characteristics
Writing hid service characteristics
Advertiser created:  Joystick  with services:  [UUID(0x1812)]
Server started
Started advertising
Central connected:  0
Returning secret: b'\x00\x8b\x8d\xef\xcf\x08\xb8\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00f\xd2\xc9|O\x07\x9b\x1ad2(>\xba u\xe9\x01\xd9\x15\xb0 \x96\xfb\xedPM\xb2\x89\x97\x97\xe2ZP\x01\xc0\x84\x88}W80e\xc8:\xcc\xd0\xba\x8b\x95\x18\x05\x00\x00\x00\x00\x00' for key (1, b'\x00\x8b\x8d\xef\xcf\x08\xb8') with type 1
Encryption update 0 1 0 1 16
MTU exchanged:  23
```

``Keys.json``:
```
[
    [1, "AIuN788IuA==", "AIuN788IuBAAAAAAAAAAAAAAAAAAAAAAZtLJfE8HmxpkMig+uiB16QHZFbAglvvtUE2yiZeX4lpQAcCEiH1XODBlyDrM0LqLlRgFAAAAAAA="], 
    [2, "AIuN788IuA==", "AIuN788IuBAAAAAAAAAAAAAAAAAAAAAAZtLJfE8HmxpkMig+uiB16QG2G2v0U7g0RduyxbyYkEEuAXWM9mEnGw7GiaeSqQYA7SMFAAAAAAA="], 
    [10, "aXJr", "2RWwIJb77VBNsomXl+JaUA=="]
]
```

Kind regards,
Heerko

closed 2025-01-27 by Heerkog