Heerkog/MicroPythonBLEHID #23: Joystick isn't picked up by the OS

dzervas, 2024-12-10:
I'm not very sure what's wrong but I tested the device in Linux and iOS

While both report the BLE device as joystick, no input is picked up and no joystick device is detected after pair & connect

I only want to use buttons (up/down/left/right and more)

I'm not even sure where to start - dmesg shows absolutely nothing and bluetooth systemd service reports:

```
profiles/input/hog-lib.c:report_ccc_written_cb() Write report characteristic descriptor failed: Attribute can't be written
```

side question: how do i know which button is which? also to add more buttons, should I change the report descriptor?

Heerkog, 2024-12-10:
Hello,

> I'm not even sure where to start - dmesg shows absolutely nothing and bluetooth systemd service reports:

It looks like your OS is writing to a characteristic which we are ignoring. The code, indeed, does not handle updates to characteristics. We could add this, however I have never encountered a case where I received an interrupt request for this. Are you receiving any unhandled IRQ events (j.e., ``_IRQ_GATTS_WRITE``)?

> side question: how do i know which button is which? also to add more buttons, should I change the report descriptor?

I'm unsure what you're asking. 

> also to add more buttons, should I change the report descriptor?

Yes. There are currently 8 buttons. More would require an additional byte in the descriptor.

> I only want to use buttons (up/down/left/right and more)

I would just translate pushing a button to the full axis movement of the joystick like [here](examples/tinypico/tinypico_dpad.py).

Kind regards,
Heerko

Heerkog, 2024-12-11:
Hi,

I just pushed an update. Please try it. When you do, please remove the device from your OS first and remove the ``keys.json`` file.

Kind regards,
Heerko

dzervas, 2024-12-13:
still won't work - I was sure to remove the bt device and keys.json
tested on both linux desktop and ios

Heerkog, 2024-12-14:
Hi,

I have done a major update for you to see what is going on. Please test it and share the output.

Kind regards,
Heerko

Heerkog, 2025-01-02:
Hi @dzervas,

Have you had a chance to test my update?

Thanks,
Heerko

mbtronics, 2025-01-17:
Hi everyone,

I'm testing your code with the keyboard example and get exactly the same error as OP.
The device pairs fine, but there is an error in the journal and no input device is added:

Jan 17 10:29:14 pegasus bluetoothd[1100]: profiles/input/hog-lib.c:report_ccc_written_cb() Write report characteristic descriptor failed: Attribute can't be written

I have extensive experience in debugging on linux, but I have no idea where to start for bluetooth specifically :-)

Heerkog, 2025-01-18:
Hi @mbtronics ,

Thank you for the insight! It seems like Linux is attempting to write to the Client Characteristic Configuration descriptor (``0x2902``). However, the keyboard example has two of these. These attribute descriptors actually should not be written to, but are set to ``ATT_F_READ_WRITE`` because of another bug.

Could you
1) confirm that this attribute is written to in the output of the keyboard example? Please include the full output.
2) Try setting the ``0x2902`` values to ``ATT_F_READ`` and see if the error persists?

Thank you!

mbtronics, 2025-01-18:
```
MPY: soft reboot
Server created
Returning secret: b"\x00sh\x14\xc5\xf8\x8c\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb8\x99\xae\xca\xf8`\x1b\xcd'\x98Sj\x10\xf3x\xe3\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00uV7\x05Lx\xc4\xbb\xe2\xb0w\x19\x7fd\x8e\xea\x05\x00\x00\x00\x00\x00" for index 0 with type 2
Returning secret: None for index 1 with type 2
Returning secret: b'\x94b\xbad$bK\x95\x81\x90\xd1r\xbe\x02G\x89' for key (10, b'irk') with type 10
Returning secret: b"\x00sh\x14\xc5\xf8\x8c\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb8\x99\xae\xca\xf8`\x1b\xcd'\x98Sj\x10\xf3x\xe3\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00uV7\x05Lx\xc4\xbb\xe2\xb0w\x19\x7fd\x8e\xea\x05\x00\x00\x00\x00\x00" for index 0 with type 2
Returning secret: None for index 1 with type 2
BLE on with public mac address b'\x08\xd1\xf9\xee\x02&'
Registering services
Writing service characteristics
Saving device information service characteristics
Saving battery service characteristics
Saving HID service characteristics
Writing service characteristics
Advertiser created:  Keyboard  with services:  [UUID(0x1812)]
Server started
Started advertising
Central connected: 0
Returning secret: b'\x00sh\x14\xc5\xf8\x8c\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb8\x99\xae\xca\xf8`\x1b\xcd\'\x98Sj\x10\xf3x\xe3\x01V3\x8f\x8f%\x95J\xbb\x8c\xca0\'\xf5\x94\'\x88\x01\x14\x1b\x93\xa93\xf6#\x8a\xe3\x9d\xffp\xd3"!\xf2\x05\x00\x00\x00\x00\x00' for key (1, b'\x00sh\x14\xc5\xf8\x8c') with type 1
Encryption update: 0 1 0 1 16
MTU exchanged: 23
Read request: HID information with value b'\x01\x01\x00\x02'
Read request: HID input report with value b'\x00\x00\x00\x00\x00\x00\x00\x00'
Read request: HID input reference with value b'\x01\x01'
Read request: HID output report with value b'\x00\x00\x00\x00\x00\x00\x00\x00'
Read request: HID output reference with value b'\x01\x02'
Read request: HID protocol mode with value b'\x01'
Read request: PnP information with value b'\x01a\xfe\x01\x00#\x01'
Read request: Battery level with value b'd'
Read request: HID input CCCD with value b'\x01'
set keys
Notify with report:  (0, 0, 161, 0, 0, 0, 0, 0)
set keys
Notify with report:  (0, 0, 161, 0, 0, 0, 0, 0)
set keys
Notify with report:  (0, 0, 161, 0, 0, 0, 0, 0)
set keys
Notify with report:  (0, 0, 161, 0, 0, 0, 0, 0)
set keys
Notify with report:  (0, 0, 161, 0, 0, 0, 0, 0)
set keys
Notify with report:  (0, 0, 161, 0, 0, 0, 0, 0)
set keys
Notify with report:  (0, 0, 161, 0, 0, 0, 0, 0)
set keys
Notify with report:  (0, 0, 161, 0, 0, 0, 0, 0)

```

mbtronics, 2025-01-18:
I changed 0x2902 to ATT_F_READ:
```
self.HIDS = (                                                                                                   
    UUID(0x1812),                                                                                              
    (
        (UUID(0x2A4A), F_READ),                                                                                 
        (UUID(0x2A4B), F_READ),                                                                                
        (UUID(0x2A4C), F_READ_WRITE_NORESPONSE),                                                               
        (UUID(0x2A4D), F_READ_NOTIFY, (                                                                    
            (UUID(0x2902), ATT_F_READ),                                                                         
            (UUID(0x2908), ATT_F_READ_WRITE),                                                                
        )),
        (UUID(0x2A4D), F_READ_WRITE, (                                                                         
            (UUID(0x2902), ATT_F_READ),                                                                      
            (UUID(0x2908), ATT_F_READ_WRITE),                                                                 
        )),
        (UUID(0x2A4E), F_READ_WRITE_NORESPONSE),                                                               
    ),
)
```

and now I get this error:
```
Jan 18 14:27:40 pegasus bluetoothd[1087]: profiles/input/hog-lib.c:ccc_read_cb() Error reading CCC value: Attribute can't be read
```

I'll try to debug it some more.

mbtronics, 2025-01-18:
I just restarted my computer, and now the keyboard is picked up by the OS with the original code, although the error is still present.

I did have some success sending chars.

```
Jan 18 15:18:49 pegasus kernel: input: Keyboard as /devices/virtual/misc/uhid/0005:FE61:0001.0017/input/input46
Jan 18 15:18:49 pegasus kernel: hid-generic 0005:FE61:0001.0017: input,hidraw3: BLUETOOTH HID v1.23 Keyboard [Keyboard] on 8c:f8:c5:14:68:73
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (II) config/udev: Adding input device Keyboard (/dev/input/event19)
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (**) Keyboard: Applying InputClass "evdev keyboard catchall"
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (**) Keyboard: Applying InputClass "libinput keyboard catchall"
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (**) Keyboard: Applying InputClass "system-keyboard"
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (II) Using input driver 'libinput' for 'Keyboard'
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (II) systemd-logind: got fd for /dev/input/event19 13:83 fd 150 paused 0
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (**) Keyboard: always reports core events
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (**) Option "Device" "/dev/input/event19"
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (II) event19 - Keyboard: is tagged by udev as: Keyboard
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (II) event19 - Keyboard: device is a keyboard
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (II) event19 - Keyboard: device removed
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (**) Option "config_info" "udev:/sys/devices/virtual/misc/uhid/0005:FE61:0001.0017/input/input46/event19"
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (II) XINPUT: Adding extended input device "Keyboard" (type: KEYBOARD, id 17)
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (**) Option "xkb_layout" "be"
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (II) event19 - Keyboard: is tagged by udev as: Keyboard
Jan 18 15:18:49 pegasus /usr/libexec/gdm-x-session[45379]: (II) event19 - Keyboard: device is a keyboard
Jan 18 15:18:50 pegasus bluetoothd[1087]: profiles/input/hog-lib.c:report_ccc_written_cb() Write report characteristic descriptor failed: Attribute can't be written
```

Heerkog, 2025-01-18:
With the original code? This is weird. Was the keyboard paired with the updated code before the reboot of the pc maybe?

It's also weird that the keyboard output is showing no write even though the OS attempting one. This means that the write request never reaches the irq function.

mbtronics, 2025-01-24:
I just started working on this again, and my device was not working again, even though I only updated main.py and not hid_services.py. Very strange.

I then changed the 0x2908 descriptors for keyboard from ATT_F_READ_WRITE to ATT_F_READ, re-uploaded hid_services.py and now it's working again! Did my OS cache something the previous time? I will be experimenting more with this later.

This is the kernel log after my latest changes:

```
Jan 24 10:04:01 pegasus kernel: input: Keyboard as /devices/virtual/misc/uhid/0005:FE61:0001.0008/input/input30
Jan 24 10:04:01 pegasus kernel: hid-generic 0005:FE61:0001.0008: input,hidraw3: BLUETOOTH HID v1.23 Keyboard [Keyboard] on 8c:f8:c5:14:68:73
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (II) config/udev: Adding input device Keyboard (/dev/input/event19)
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (**) Keyboard: Applying InputClass "evdev keyboard catchall"
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (**) Keyboard: Applying InputClass "libinput keyboard catchall"
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (**) Keyboard: Applying InputClass "system-keyboard"
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (II) Using input driver 'libinput' for 'Keyboard'
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (II) systemd-logind: got fd for /dev/input/event19 13:83 fd 102 paused 0
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (**) Keyboard: always reports core events
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (**) Option "Device" "/dev/input/event19"
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (II) event19 - Keyboard: is tagged by udev as: Keyboard
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (II) event19 - Keyboard: device is a keyboard
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (II) event19 - Keyboard: device removed
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (**) Option "config_info" "udev:/sys/devices/virtual/misc/uhid/0005:FE61:0001.0008/input/input30/event19"
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (II) XINPUT: Adding extended input device "Keyboard" (type: KEYBOARD, id 17)
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (**) Option "xkb_layout" "be"
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (II) event19 - Keyboard: is tagged by udev as: Keyboard
Jan 24 10:04:01 pegasus /usr/libexec/gdm-x-session[3313]: (II) event19 - Keyboard: device is a keyboard
Jan 24 10:04:01 pegasus bluetoothd[1095]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Attribute can't be read
Jan 24 10:04:02 pegasus bluetoothd[1095]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Attribute can't be read
```

Heerkog, 2025-01-24:
Could you try setting 0x2908 to ATT_F_READ_WRITE and 0x2908 to ATT_F_READ?

Heerkog, 2025-01-24:
If that doesn't work, make them both ATT_F_READ, please

Heerkog, 2025-01-24:
@mbtronics I just pushed a possible bug fix. Could you please try it?

mbtronics, 2025-01-25:
I just tried the master version of hid_services.py:
- forget device
- send new version
- connect with device

No keyboard is detected (although the device is sending chars) and the following is in the linux logs:

```
Jan 25 14:20:14 pegasus bluetoothd[1096]: src/device.c:set_wake_allowed_complete() Set device flags return status: Invalid Parameters
Jan 25 14:20:14 pegasus kernel: Bluetooth: hci0: Bad flag given (0x1) vs supported (0x0)
Jan 25 14:20:17 pegasus bluetoothd[1096]: profiles/input/hog-lib.c:report_ccc_written_cb() Write report characteristic descriptor failed: Attribute can't be written
```

Log on the device looks like this:

```
Server created
Returning secret: b'\x00sh\x14\xc5\xf8\x8c\x00\x00\x00\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1fF\x04m\xf0\xb3\xba\xf1\xaf\xa3]\xb5\x9f\x99\xbe\x98\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x000\xb7}\x19\xff\x95\xb9\xb74\x95\xb7\xa1Z\xfc\xb3m\x01\x00\x00\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00' for index 0 with type 2
Returning secret: None for index 1 with type 2
Returning secret: b'\xa8h\xeax[13\xefb\xdbY\xd8\x90\x1d"`' for key (10, b'irk') with type 10
Returning secret: b'\x00sh\x14\xc5\xf8\x8c\x00\x00\x00\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1fF\x04m\xf0\xb3\xba\xf1\xaf\xa3]\xb5\x9f\x99\xbe\x98\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x000\xb7}\x19\xff\x95\xb9\xb74\x95\xb7\xa1Z\xfc\xb3m\x01\x00\x00\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00' for index 0 with type 2
Returning secret: None for index 1 with type 2
BLE on with public mac address b'\x08\xd1\xf9\xee\x02&'
Registering services
Writing service characteristics
Saving device information service characteristics
Saving battery service characteristics
Saving HID service characteristics
Writing service characteristics
Advertiser created:  Keyboard  with services:  [UUID(0x1812)]
Server started
start connection
Started advertising
Central connected: 0
Returning secret: b'\x00sh\x14\xc5\xf8\x8c\x00\x00\x00\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1fF\x04m\xf0\xb3\xba\xf1\xaf\xa3]\xb5\x9f\x99\xbe\x98\x01\xa8h\xeax[13\xefb\xdbY\xd8\x90\x1d"`\x01\x0f\xf1\x15\x0f\xa1VC\xa7Te\xf0\xdf\xda\x13L\xfa\x01\x00\x00\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00' for key (1, b'\x00sh\x14\xc5\xf8\x8c') with type 1
Encryption update: 0 1 0 1 16
MTU exchanged: 23
Read request: PnP information with value b'\x01a\xfe\x01\x00#\x01'
Read request: HID information with value b'\x01\x01\x00\x02'
Read request: HID input reference with value b'\x01\x01'
Read request: HID output reference with value b'\x01\x02'
Read request: HID protocol mode with value b'\x01'
Read request: PnP information with value b'\x01a\xfe\x01\x00#\x01'
Read request: Battery level with value b'd'
Read request: HID input CCCD with value b'\x01'
Notify with report:  (0, 0, 24, 0, 0, 0, 0, 0)
```

mbtronics, 2025-01-25:
Do you have any information regarding this "allow write because MicroPython v1.20+ bug"?

This appears to be the cause: with 0x2908 on read/write the device appears to be pairing but not working, with 0x2908 on read only it appears to be working but not pairing.

Heerkog, 2025-01-25:
My newest version should fix that bug. Also we should try cccd with value
0x00 or removing them entirely because the stack should add those
automaticaly.

Heerkog, 2025-01-26:
I pushed a version without cccd.

mbtronics, 2025-01-27:
Nice, latest version from master seems to work much better. Pairing works and I'm receiving chars immediately after that.
There are still some errors in the linux logs on pairing, but apparently not fatal ones:

```
**Jan 27 09:43:37 pegasus kernel: Bluetooth: hci0: Bad flag given (0x1) vs supported (0x0)**
**Jan 27 09:43:37 pegasus bluetoothd[1096]: src/device.c:set_wake_allowed_complete() Set device flags return status: Invalid Parameters**
Jan 27 09:43:40 pegasus kernel: input: Keyboard as /devices/virtual/misc/uhid/0005:FE61:0001.0044/input/input90
Jan 27 09:43:40 pegasus kernel: hid-generic 0005:FE61:0001.0044: input,hidraw3: BLUETOOTH HID v1.23 Keyboard [Keyboard] on 8c:f8:c5:14:68:73
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (II) config/udev: Adding input device Keyboard (/dev/input/event19)
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (**) Keyboard: Applying InputClass "evdev keyboard catchall"
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (**) Keyboard: Applying InputClass "libinput keyboard catchall"
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (**) Keyboard: Applying InputClass "system-keyboard"
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (II) Using input driver 'libinput' for 'Keyboard'
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (II) systemd-logind: got fd for /dev/input/event19 13:83 fd 496 paused 0
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (**) Keyboard: always reports core events
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (**) Option "Device" "/dev/input/event19"
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (II) event19 - Keyboard: is tagged by udev as: Keyboard
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (II) event19 - Keyboard: device is a keyboard
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (II) event19 - Keyboard: device removed
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (**) Option "config_info" "udev:/sys/devices/virtual/misc/uhid/0005:FE61:0001.0044/input/input90/event19"
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (II) XINPUT: Adding extended input device "Keyboard" (type: KEYBOARD, id 17)
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (**) Option "xkb_layout" "be"
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (II) event19 - Keyboard: is tagged by udev as: Keyboard
Jan 27 09:43:41 pegasus /usr/libexec/gdm-x-session[3301]: (II) event19 - Keyboard: device is a keyboard
```

This is how it looks on second connect (no errors):
```
Jan 27 09:46:33 pegasus kernel: input: Keyboard as /devices/virtual/misc/uhid/0005:FE61:0001.0045/input/input91
Jan 27 09:46:33 pegasus kernel: hid-generic 0005:FE61:0001.0045: input,hidraw3: BLUETOOTH HID v1.23 Keyboard [Keyboard] on 8c:f8:c5:14:68:73
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (II) config/udev: Adding input device Keyboard (/dev/input/event19)
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (**) Keyboard: Applying InputClass "evdev keyboard catchall"
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (**) Keyboard: Applying InputClass "libinput keyboard catchall"
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (**) Keyboard: Applying InputClass "system-keyboard"
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (II) Using input driver 'libinput' for 'Keyboard'
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (II) systemd-logind: got fd for /dev/input/event19 13:83 fd 511 paused 0
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (**) Keyboard: always reports core events
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (**) Option "Device" "/dev/input/event19"
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (II) event19 - Keyboard: is tagged by udev as: Keyboard
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (II) event19 - Keyboard: device is a keyboard
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (II) event19 - Keyboard: device removed
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (**) Option "config_info" "udev:/sys/devices/virtual/misc/uhid/0005:FE61:0001.0045/input/input91/event19"
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (II) XINPUT: Adding extended input device "Keyboard" (type: KEYBOARD, id 17)
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (**) Option "xkb_layout" "be"
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (II) event19 - Keyboard: is tagged by udev as: Keyboard
Jan 27 09:46:34 pegasus /usr/libexec/gdm-x-session[3301]: (II) event19 - Keyboard: device is a keyboard
```

Heerkog, 2025-01-27:
Excellent!

The latest push now includes two more services. One of which sets the value of the non-critical bug in your first connect.

closed 2025-01-27 by Heerkog